penk/penkesu #2: Os

gedw99, 2022-02-14:
Have you decided on an OS ?

gokrazy might be a good fit for super low resource device . I run it 

https://github.com/gokrazy

for apps you can use Fyne or gioui

https://github.com/fyne-io/fyne

https://github.com/gioui

This makes it very lean and easy for you and others to build apps for it .

gedw99, 2022-02-14:
Also AdGuard ad blocker is programmed in go so if you wanted to add a web browser it would be relatively easy to integrate AdGuard into it . 

gioui is similar to QT but

gedw99, 2022-02-14:
https://gokrazy.org/platforms/#raspberry-pi-zero-2-w-work-in-progress

penk, 2022-02-14:
@gedw99 thank you for your comments! Right now we are using Raspberry Pi OS Bullseye since it's working out-of-the-box. 

(btw I'm one of the author of cutiepi-shell so I'm flattered ;-)

closed 2022-02-14 by penk

gedw99, 2022-02-15:
no worries.

gedw99, 2022-02-15:
@penk when can i get my hands on the hardware ?