penk/penkesu #4: Support Radxa Zero

hipboi, 2022-02-15:
Raspberry Pi Zero 2 W ram is only 512MB, which is not enough for desktop usage. Could you support [Radxa Zero](https://wiki.radxa.com/Zero) 4GB ram model?

penk, 2022-02-15:
That looks cool, do you know how do I get one?

Drizzt321, 2022-02-15:
That's interesting. The [buy page](https://wiki.radxa.com/Buy) shows [Seeed](https://www.seeedstudio.com/ROCK-Pi-4-c-1323.html). Don't see it there though, strangely. Does seem to be on [ALLNET China](https://shop.allnetchina.cn/collections/frontpage/products/copy-of-radxa-zero) though, and also includes options for up to 128GB eMMC which seems quite interesting. Not all that pricey for that, although not sure on shipping. 

Still supports mSD cards, and USB-C instead of micro-B for power & data, and micro HDMI connector vs mini HDMI on the RPi Zero 2W. So still pretty standard, but probably needing slightly different cables/adapters. I haven't looked close, but the 40-pin header I bet is compatible.

hipboi, 2022-02-16:
@penk  We can just ship one free sample to you. Please [send me](mailto:tom@radxa.com) your shipping address.

RicInNewMexico, 2022-02-16:
> @penk We can just ship one free sample to you. Please [send me](mailto:tom@radxa.com) your shipping address.

Now that's community support! I was just going to comment and list which distributors carry the Radxa Zero.. just ordered a 4GB ram / 32GB eMMC for my penkesu build last night

cristicbz, 2022-04-08:
@hipboi I tried to get this to work for a couple of evenings, but wasn't able to get the screen to work. Tried a custom modeline, kernel params, but no luck. Seems like the modesetting driver doesn't support the odd resolution (400x1280)? 

Someone else also struggled by looks of it: https://forum.radxa.com/t/waveshare-7-9in-touchscreen-support/7555

I'm by no means an expert, so I'm sure someone else could make it work. Might need a kernel patch

hipboi, 2022-04-17:
@cristicbz somehow I missed this thread, @RadxaYuntian will follow up this issue. I believe we need some custom edid for this panel.

markcordova, 2022-05-22:
@hipboi this issue is now closed, does this mean that the radxa zero is tested to be working fine in this setup? thanks

2disbetter, 2022-05-22:
@markcordova unfortunately a radxa Zero was not available. Instead of leaving the issue open and hanging, it is just closed until one can be attained. I personally think a Radxa would be excellent in there. I'm a little worried about clearances, but it should be fine.

Drizzt321, 2022-05-24:
@2disbetter I got my hands on a Radxa Zero, haven't done anything yet. I can lend it to you to do the testing you need, and then send it back.

2disbetter, 2022-06-10:
Thanks Drizzt, I was able to get one too. I'm tinkering with it now. Just trying to get the waveshare display to work. I really want to check the power requirements and see how the thermals are. Will post again when I have something to report.

kariudo, 2022-06-10:
> Thanks Drizzt, I was able to get one too. I'm tinkering with it now. Just trying to get the waveshare display to work. I really want to check the power requirements and see how the thermals are. Will post again when I have something to report.

Should probably re-open the issue, if its in progress. I'd be interested in this too, especially with hardware shortages for things like zero w 2.

2disbetter, 2022-06-10:
right now the display not working is the main issue. Still working on it, but would be very happy if we could get a comment from hipboi about the driver and the waveshare screen used with the Penkesu.

jcbergerson, 2022-07-24:
I also am curious about the radxa Zero/Zero 2 @hipboi for the penkesu platform. The video driver and support of the waveshare screen would be awesome! The future collaboration of the open @penk pocketPC and the possibilities of the Radxa Zero platform as it’s heartbeat would be cutting edge magical!

RadxaYuntian, 2022-07-25:
So the preliminary investigation shows this is likely a driver issue. The resolution for the screen is actually 400x1280 as that's the default display orientation. In meson driver, there are [predefined resolutions](https://github.com/analogdevicesinc/linux/blob/a34dd103e47072bad06d513e356b2cf5df0f71c4/drivers/gpu/drm/meson/meson_venc.c#L821) and custom resolutions. Any custom resolutions with horizontal resolution less than 640, or vertical resolution greater than 1200 [is rejected](https://github.com/analogdevicesinc/linux/blob/master/drivers/gpu/drm/meson/meson_venc.c#L869=) by the driver.

We might be able to add a custom resolution in the predefined list. Will need more investigation to solve this issue.

2disbetter, 2022-11-28:
> We might be able to add a custom resolution in the predefined list. Will need more investigation to solve this issue.

Any luck on this? I have since been able to source a battery charger that can output 5v at a max of 3A. I would really love to try the Radxa with a USB hub and see if I can get the display and the touchscreen working. It would be one significant upgrade to the Penkesu.

RadxaYuntian, 2022-11-29:
The driver limitation [was lifted](https://github.com/radxa/kernel/pull/26) so display should work now. The touch screen IIRC is USB based? In that case it should work as well.

2disbetter, 2023-05-09:
Thank you, we will work on adding this to the build. This would represent a significant bump in performance.

closed 2023-05-09 by 2disbetter

markcordova, 2023-06-07:
@2disbetter I can confirm that the display now works with the Radxa Zero, and with a USB hub the touchscreen works as well.