penk/penkesu #11: Keyboard connection

gopher333, 2022-06-28:
Hi, I would appreciate a little more detail about the hardware configuration of the keyboard. How do I attach the Arduino Mini to the Keyboard (Pin Numbers)? I suppose the USB Port has to be to the outside of the PCB, but in which direction? USB side to the PCB or away from it? A picture would do...also the connection of the 4-pin usb connector would be helpfull (PINS?)
Tks is advance?

2disbetter, 2022-06-28:
[image]
The micro usb port goes facing the Keyboard PCB. This picture should show you the pins. Be careful that nothing is touching metal on metal there. I actually put a small piece of kapton tape in between just to make sure.

gopher333, 2022-06-29:
Thanks a lot for thte immediate answer, that does it!

closed 2022-06-29 by gopher333