penk/penkesu #12: Touch Display usable?

gopher333, 2022-06-29:
Dear Penk, my Penkesu is now almost finished! Thanks for the work, and most for sharing it! Just a question; As far as the used display is a touch display, is there any argument against using it that way? There is still an usb Port on the RasPi available...

cap-steve, 2022-12-12:
wondering if this ever got tested.
i've been looking at various usb hats and wonder if it's worthwhile to pursue, even at the expense of battery life.

markcordova, 2023-06-16:
I use an internal 4-port USB hub to connect the touchscreen and the keyboard (search for fe1.1s USB hub if interested).

2disbetter, 2023-09-27:
> I use an internal 4-port USB hub to connect the touchscreen and the keyboard (search for fe1.1s USB hub if interested).

Could you share a picture of how you have it configured internally on your Penkesu? Is power supply sufficient?

Steuv1871, 2024-03-02:
Hey ! I got the touchscreen to work with a usb hub + OTG adapter on Pi Zero mini USB.  
I'm using a [pi-juice zero hat](https://uk.pi-supply.com/products/pijuice-zero) and it supply enought power.  
More than that, the display don't work unless I plug the usb cable (power over hdmi is not enought).

It's a bit difficult to touch things too close from the border due to the screen panel thickness. But it's usable enought for desktop WM like xfce.

But be carefull with the pi-juice zero hat, it's too thick for the Penkesu original design, I had to modify the STLs.