penk/penkesu #14: Booting directly to cool-retro-term console

Kingeorg, 2022-11-30:
Hello. 
-1st. I must thank you for your project and the inspiration provided, I was looking for a portable terminal and this is the perfect size.
-2nd. I successfully completed my Penkesu! It works perfectly, albeit with some modifications (using instead a 30% gherkin keyboard) and a custom made LEGO case, it is beautiful!
-3rd. Here's my question. How can I boot directly into cool-retro-terminal (the terminal emulator you use with your Main Board Terminal project)? I know this is a software issue, but I figured you would know as you referr to it in your other project. Hardware-wise, the Raspberry Pi Zero 2W I'm using does run cool-retro-term at a decent speed when you adjust the performance from within the program, my issue is that I'd like to run it as closest to "bare metal" as possible, with the minimum requirements, and directly as possible. I managed to boot to cool-retro-term but only after the Raspbian desktop environment loads (it takes a while and the transition is ugly). Unfortunately cool-retro-term seems to need some kind of Window Manager, so It cannot boot directly from the terminal before the desktop environment starts. I tried loading have spent a lot of time trying to figure it out, whitout getting good results. What do you suggest?

penk, 2022-12-05:
Hi @Kingeorg thank you for the ticket, that's awesome! If you have any pictures of your build please do let me know, would love to see it! :-) 

For booting directly into cool retro term, I think the easiest way is to add an `.desktop` file under `/etc/xdg/autostart` so it'll launch automatically when the desktop starts -- it is heavier (with Xorg and mutter window manager and other layers). 

There's another way you can probably try, stop your Xorg with `systemctl stop lightdm` and starting cool-retro-term with `-platform eglfs` parameter, I haven't tested this but it should work without any display server or window manager. 
Let me know how it goes, you'll probably have to grand your user account with permission to draw directly:

```
sudo gpasswd -a pi render
sudo gpasswd -a pi dialout
sudo gpasswd -a pi input
```

You can find more info about EGLFS here: https://doc.qt.io/qt-5/embedded-linux.html#eglfs 

Let me know how it goes!

Kingeorg, 2023-01-13:
Thank you. I'll give it a try now that I have time and let you know how it goes.
In the meantime, I'd like to share what I made inspired by your Penkesu project:
https://www.reddit.com/r/cyberDeck/comments/zfbioy/my_wallnut_lego_cyberdeck/

2disbetter, 2023-04-30:
Seems solution was given, or a possible, and as such closing this one. Thanks!

closed 2023-04-30 by 2disbetter