penk/penkesu #16: Need help to rotate the screen

CJ-eng, 2023-01-26:
Hi Penk,

I've hooked up everything minus the keyboard (still waiting for PCB). However, for testing purpose, I hook up regular keyboard to Pi Zero's USB port. So everything works nicely except the default Waveshare screen is vertical. I edit the config.txt file as per Waveshare's Wiki page but it is not working :-(

Please let me know how to get the screen to rotate. Thanks!!!

closed 2023-01-26 by CJ-eng

penk, 2023-01-26:
For the record it was set with xrandr commands. :-)

CJ-eng, 2023-01-26:
Thanks for your reply. I have to delete everything else in the config.txt file and then the rotate command works. But I will try your command as well. I am enjoying the pencil case PC very much!