penk/penkesu #17: Where are the - and + key?

CJ-eng, 2023-04-30:
Hi,

I have used your key map and I can't find the - and + key. I've searched  the "raise" layer or the "lower" layer. Please advises. Thanks.

2disbetter, 2023-04-30:
The minus (-) can be accessed via lower + h, and the plus (+) can be accessed via raise + j. Not that intuitive for sure but a small keyboard requires some compromises. BTW, is this CJ Engineering from Youtube, of Framework fame? If so hi, from the same person on the Framework forums.

CJ-eng, 2023-04-30:
Thank you so much for fast response. Yes indeed the key is there, I was testing the keymap by mechanical keyboard tester online and somehow the key is not registered. But all is well now.

I am not familiar with the person you mentioned. The name is just coincidental.

Thanks and keep up the good work!

closed 2023-04-30 by 2disbetter